TVavrinec/Servio-PCB #4: Attach debugger directly to the board.

koniarik, 2024-03-10:
[image]

We designed this to connect the debugger via cable to the board being debugged. On prototype I realized that this is just waste of space, as we could just put female pin headers on the debug board and connect the debugger directly. 

P.S: If we minimize space required by the whole setup it's easier to setup "small" testjigs. Which is necessary to be able to have a lot of them.

TVavrinec, 2024-03-10:
Well, it's a good idea, it's just a matter of soldering the female pin headers, no need to change the PCB.

koniarik, 2024-03-10:
> Well, it's a good idea, it's just a matter of soldering the female pin headers, no need to change the PCB.

There is no need for mirroring or something? coool

TVavrinec, 2024-03-10:
Yes, the cabin doesn't change the position of the signals.
[image]
[image]

closed 2025-02-27 by koniarik